RomanLut/hx_espnow_rc #6: [Questions] Is 50Hz the max rate for ESPNow?

human890209, 2022-06-24:
Hi,
I'm wondering if 50Hz is the max update rate for ESPNow? I know LR is slow. If disabled LR mode could ESP32 be updated faster than 50Hz? What's the bottleneck of this system? ESP01?

RomanLut, 2022-06-24:
You can increase packet rate 2x if you are not going to send a lot of telemetry:
https://github.com/RomanLut/hx_espnow_rc/blob/main/doc/rfpower/rfpower.md

In normal mode rate is 1MBit/s and in LR I think 500KBit. Every wifi packet contains relatively large 802.11 header. 50Hz is reasonable update rate to have communication with telemetry at this transmission rates.

Note that this is still Wifi communication; channel timing is still shared with wifi devices on the same channel.

I think BW_HT20 does not have effect on LR mode and on WIFI_PROTOCOL_11B mode.  It is for WIFI_PROTOCOL_11G mode.

LR could be slowed down more and potentionally have higher range with WIFI_PHY_RATE_LORA_250K. Unfortunatelly it is not possible to enable 250K rate from Arduino SDK.

RomanLut, 2022-06-24:
Well, actually I do not know with rate is default for LR mode: 250k or 500k due to luck of docmentation:
https://www.esp32.com/viewtopic.php?t=24212

closed 2022-06-24 by human890209