RomanLut/hx_espnow_rc #7: Cannot compile external module

geofrancis, 2022-07-28:
I am trying to compile the external module but i keep getting bluetooth errors when compiling, the esp-01 Sbus reciver compiled and flashed ok.

`Compiling .pio\build\esp32doit-devkit-v1\liba50\ESP32 BLE Arduino\BLECharacteristic.cpp.o
In file included from C:/Users/Geo/Documents/CODE/hx_espnow_rc-0.1/lib/hx_espnow_rc/HX_ESPNOW_RC_Base.h:5,
                 from C:/Users/Geo/Documents/CODE/hx_espnow_rc-0.1/lib/hx_espnow_rc/HX_ESPNOW_RC_Master.h:5,
                 from src/main.cpp:8:
C:/Users/Geo/Documents/CODE/hx_espnow_rc-0.1/lib/hx_espnow_rc/HX_ESPNOW_RC_Common.h:26:10: fatal error: esp_wifi_internal.h: No such file or directory

***************************************************************************
* Looking for esp_wifi_internal.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_wifi_internal.h"
* Web  > https://registry.platformio.org/search?q=header:esp_wifi_internal.h
*
***************************************************************************

 #include "esp_wifi_internal.h"
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32doit-devkit-v1\src\main.cpp.o] Error 1
In file included from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLECharacteristic.h:19,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEDescriptor.h:14,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLE2902.h:13,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLE2902.cpp:15:
.pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
  Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);
                            ^~~~~~~~~~~~~~
In file included from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLECharacteristic.h:19,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEDescriptor.h:14,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLE2904.h:13,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLE2904.cpp:15:
.pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
  Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);
                            ^~~~~~~~~~~~~~
*** [.pio\build\esp32doit-devkit-v1\liba50\ESP32 BLE Arduino\BLE2902.cpp.o] Error 1
*** [.pio\build\esp32doit-devkit-v1\liba50\ESP32 BLE Arduino\BLE2904.cpp.o] Error 1
In file included from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEAdvertising.h:15,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEAdvertising.cpp:21:
.pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
  Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);
                            ^~~~~~~~~~~~~~
In file included from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLERemoteDescriptor.h:18,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLERemoteCharacteristic.h:18,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLERemoteService.h:16,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEClient.h:19,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEScan.h:17,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEAdvertisedDevice.h:17,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEAdvertisedDevice.cpp:17:       
.pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
  Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);
                            ^~~~~~~~~~~~~~
*** [.pio\build\esp32doit-devkit-v1\liba50\ESP32 BLE Arduino\BLEAdvertising.cpp.o] Error 1
*** [.pio\build\esp32doit-devkit-v1\liba50\ESP32 BLE Arduino\BLEAdvertisedDevice.cpp.o] Error 1
In file included from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEDescriptor.h:16,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLECharacteristic.h:17,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLECharacteristic.cpp:15:
.pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
  Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);
                            ^~~~~~~~~~~~~~
*** [.pio\build\esp32doit-devkit-v1\liba50\ESP32 BLE Arduino\BLECharacteristic.cpp.o] Error 1
========================================= [FAILED] Took 9.29 seconds =========================================

 *  The terminal process "C:\Users\Geo\.platformio\penv\Scripts\platformio.exe 'debug'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

`

geofrancis, 2022-07-28:
thank you. I will get back to you.

geofrancis, 2022-07-28:
could you upload a binary so I can test hardware while I sort out the software issues.

geofrancis, 2022-07-28:
can you post a binary, I have it compiling but im getting flash errors and its just constantly resetting.

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

RomanLut, 2022-07-28:
Did you upload filesystem after flashing?

RomanLut, 2022-07-28:
> #include <ESP-FTP-Server-Lib.h>

peterus/ESP-FTP-Server-Lib@^0.9.9

This dependecy is listed in platformio.ini
It should be installed automatically.

geofrancis, 2022-07-28:
upload file system?

RomanLut, 2022-07-28:
[image]

There is \data folder in the project with configuration and data files.
After flashing, upload it using (2)

geofrancis, 2022-07-28:
ok i dont that and I think its working its now showing as Waiting for CH16 profile on the serial port.

RomanLut, 2022-07-28:
Yes it is waiting for CH16 value to select profile

https://github.com/RomanLut/hx_espnow_rc/blob/main/doc/tx_external_module.md#see-next

geofrancis, 2022-07-28:
ok how do i bind it to a recevier?

geofrancis, 2022-07-28:
you should make a post over at https://discuss.ardupilot.org/c/blog/ Im sure there are a lot of people there that would be very interested in a high bandwidth mavlink RC system.

RomanLut, 2022-07-28:
> ok how do i bind it to a receiver? ok nevermind i think its not doing anything until it gets that channel 16 signal from a transmitter.

There is no bind process. You just need to have the same

	"espnow_channel": 3,
	"espnow_key": 0,
	"espnow_long_range_mode": false,

in profile.json and in config.h of receiver.

geofrancis, 2022-07-28:
It looks like it's all working, apart from I don't have a transistor to invert the SBUS input into the transmitter module. so i have ordered some and they should get here in a day or so , so im going to work on some 3d printed modules for my taranis x9d.

geofrancis, 2022-07-29:
do you support the esp8285? its something like 250mw of rf compared to the <100mw you get with the rest of the modules.

RomanLut, 2022-07-29:
https://www.espressif.com/sites/default/files/documentation/0a-esp8285_datasheet_en.pdf

802.11 b: +19 dBm

20dbm => 100mW

geofrancis, 2022-07-29:
25db with PA 
https://www.aliexpress.com/item/32846843421.html

RomanLut, 2022-07-29:
> I just noticed the esp8266MOD also has the PA https://www.aliexpress.com/item/32675165049.html

It has nothing to do with real output power, no external PA there, https://www.esp8266.com/viewtopic.php?p=938
Real output is 20 dbm to pass certification.

You may try to compile for 8285 by changing project settings.

RomanLut, 2022-07-29:
https://fccid.io/2A54N-ESP8266/Internal-Photos/Internal-Photos-5839753

geofrancis, 2022-07-29:
cool. il just stick to the esp32 and 8266.

geofrancis, 2022-07-29:
I have been busy, flashed an esp01, a pair of wemos D1 modules, one as mavlink and one as SBUS. when i was digging though my box of modules i found this pixhawk wifi adapter, you can get them for around 10usd, the d1 mini mavlink firmware flashes to it fine.

https://i.loli.net/2021/10/12/S3PvGlRwiMcLN1X.jpg

AlksSAV, 2022-07-29:
> I have been busy, flashed an esp01, a pair of wemos D1 modules, one as mavlink and one as SBUS. when i was digging though my box of modules i found this pixhawk wifi adapter, you can get them for around 10usd, the d1 mini mavlink firmware flashes to it fine.
> 
> 
> 
> 
> 
> https://www.aliexpress.com/item/1005003434945109.html
> 
> 
> 
> this looks really interesting, if it ever goes on sale 
> 
> https://www.crowdsupply.com/
bison-science/esp32-m1-reach-out#products

A module with aliexpress, this is a regular module stitched with this firmware.

geofrancis, 2022-07-29:
Yes i have that firmware on one of my d1 minis to connect it to my wifi network.

geofrancis, 2022-07-29:
@RomanLut Have you thought about using the OpenTX multimodule interface to configure the modules and profiles?

RomanLut, 2022-07-29:
Looks like a lot of work to get the the same functionality which is implemented already.
As for profiles configuration, a configurator web page hosted on module  would have much better user experience and also serve well for DIY controller. So far plain text editor web page is implemented only. I plan to implement full featured configurator some time later. Currenlty focused on INAV HITL project.

geofrancis, 2022-07-29:
nearly have a JR module built.
https://discuss.ardupilot.org/uploads/default/original/3X/0/6/069cfbb748416ce63a7c35e0535b5b147b3d7243.jpeg
https://discuss.ardupilot.org/uploads/default/original/3X/0/3/0375831f110d4982bf33ef643c25b0c177f22332.jpeg

closed 2022-08-18 by geofrancis

geofrancis, 2022-09-20:
unrelated but this looks like a good receiver 
https://www.aliexpress.com/item/4001054031743.html

geofrancis, 2022-09-26:
https://www.aliexpress.com/item/1005004604063847.html