RomanLut/hx_espnow_rc #10: Unable to build for  TX External Module

devanesanandrews, 2022-11-12:
I tried to External TX Module

Compiling .pio\build\esp32doit-devkit-v1\lib575\ESP32 BLE Arduino\BLEDescriptor.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib575\ESP32 BLE Arduino\BLEDescriptorMap.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib575\ESP32 BLE Arduino\BLEDevice.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib575\ESP32 BLE Arduino\BLEHIDDevice.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib575\ESP32 BLE Arduino\BLERemoteCharacteristic.cpp.o
In file included from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLECharacteristic.h:19,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEDescriptor.h:14,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLE2902.h:13,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLE2902.cpp:15:
.pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
  Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);
                            ^~~~~~~~~~~~~~
In file included from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLECharacteristic.h:19,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEDescriptor.h:14,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLE2904.h:13,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLE2904.cpp:15:
.pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
  Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);
                            ^~~~~~~~~~~~~~
*** [.pio\build\esp32doit-devkit-v1\lib575\ESP32 BLE Arduino\BLE2902.cpp.o] Error 1
*** [.pio\build\esp32doit-devkit-v1\lib575\ESP32 BLE Arduino\BLE2904.cpp.o] Error 1
In file included from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEAdvertising.h:15,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEAdvertising.cpp:21:
.pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
  Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);
                            ^~~~~~~~~~~~~~
In file included from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLERemoteDescriptor.h:18,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLERemoteCharacteristic.h:18,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLERemoteService.h:16,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEClient.h:19,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEScan.h:17,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEAdvertisedDevice.h:17,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEAdvertisedDevice.cpp:17:
.pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
  Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);
                            ^~~~~~~~~~~~~~
*** [.pio\build\esp32doit-devkit-v1\lib575\ESP32 BLE Arduino\BLEAdvertising.cpp.o] Error 1
In file included from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEDescriptor.h:16,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLECharacteristic.h:17,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEHIDDevice.h:14,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEHIDDevice.cpp:10:
.pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
  Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);
                            ^~~~~~~~~~~~~~
In file included from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLERemoteDescriptor.h:18,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLERemoteCharacteristic.h:18,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLERemoteService.h:16,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEClient.h:19,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEClient.cpp:13:
.pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
  Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);
                            ^~~~~~~~~~~~~~
*** [.pio\build\esp32doit-devkit-v1\lib575\ESP32 BLE Arduino\BLEHIDDevice.cpp.o] Error 1
In file included from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEDescriptor.h:16,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLECharacteristic.h:17,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEDescriptorMap.cpp:11:
.pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
  Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);
                            ^~~~~~~~~~~~~~
*** [.pio\build\esp32doit-devkit-v1\lib575\ESP32 BLE Arduino\BLEAdvertisedDevice.cpp.o] Error 1
In file included from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEDescriptor.h:16,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLECharacteristic.h:17,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEService.h:15,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLECharacteristicMap.cpp:11:
.pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
  Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);
                            ^~~~~~~~~~~~~~
In file included from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEDescriptor.h:16,
[... 28 lines cut ...]
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEDevice.h:18,
                 from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEDevice.cpp:26:
.pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
  Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);
                            ^~~~~~~~~~~~~~
*** [.pio\build\esp32doit-devkit-v1\lib575\ESP32 BLE Arduino\BLEClient.cpp.o] Error 1
*** [.pio\build\esp32doit-devkit-v1\lib575\ESP32 BLE Arduino\BLERemoteCharacteristic.cpp.o] Error 1
.pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEDevice.cpp: In static member function 'static void BLEDevice::whiteListAdd(BLEAddress)':
.pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEDevice.cpp:490:75: error: too few arguments to function 'esp_err_t esp_ble_gap_update_whitelist(bool, uint8_t*, esp_ble_wl_addr_type_t)'
  esp_err_t errRc = esp_ble_gap_update_whitelist(true, *address.getNative());  // True to add an entry.
                                                                           ^
In file included from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEDevice.cpp:17:
C:/Users/Captain kumaran/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/bt/host/bluedroid/api/include/api/esp_gap_ble_api.h:1478:11: note: declared here
 esp_err_t esp_ble_gap_update_whitelist(bool add_remove, esp_bd_addr_t remote_bda, esp_ble_wl_addr_type_t wl_addr_type);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEDevice.cpp: In static member function 'static void BLEDevice::whiteListRemove(BLEAddress)':
.pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEDevice.cpp:504:76: error: too few arguments to function 'esp_err_t esp_ble_gap_update_whitelist(bool, uint8_t*, esp_ble_wl_addr_type_t)'
  esp_err_t errRc = esp_ble_gap_update_whitelist(false, *address.getNative());  // False to remove an entry.
                                                                            ^
In file included from .pio/libdeps/esp32doit-devkit-v1/ESP32 BLE Arduino/src/BLEDevice.cpp:17:
C:/Users/Captain kumaran/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/bt/host/bluedroid/api/include/api/esp_gap_ble_api.h:1478:11: note: declared here
 esp_err_t esp_ble_gap_update_whitelist(bool add_remove, esp_bd_addr_t remote_bda, esp_ble_wl_addr_type_t wl_addr_type);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
*** [.pio\build\esp32doit-devkit-v1\lib575\ESP32 BLE Arduino\BLEDevice.cpp.o] Error 1
========================================================================================== [FAILED] Took 8.42 seconds ==========================================================================================

RomanLut, 2022-11-13:
Please update from main branch. Should be fixed.

devanesanandrews, 2022-11-13:
Cloning the Main Branch Solved the issue.

@RomanLut leave this issue here for a while, so it can be helpful for someone

closed 2023-08-28 by RomanLut