RomanLut/hx_espnow_rc #13: ESP32 not connect ESP01

thanhhuyen6sdghsjdgq, 2022-12-21:
I have tried using ESp32 as controller and loading code rx_esp01_sbus for esp01 but i am having problem that they cannot connect to each other

RomanLut, 2022-12-21:
What code is flashed to esp32?

thanhhuyen6sdghsjdgq, 2022-12-21:
I flashed it with the code tx_diy_controller

RomanLut, 2022-12-21:
Please provide  teminal output from esp32.

thanhhuyen6sdghsjdgq, 2022-12-21:
esp32 got wifi but esp01 failed to connect to wifi "hxrct"

RomanLut, 2022-12-21:
"espnow_long_range_mode": false,
- should be false in active profile in order to allow esp01 to connect to esp32.

nopi890, 2024-01-05:
Saya memflash esp32 dengan kode tx DIY controller tapi muncul error' xtensa/hal.h no such file or directory

RomanLut, 2024-01-05:
There is probably some problem with Platformio/SDK installation
https://community.platformio.org/t/after-update-compile-error-in-library/29625/3

nopi890, 2024-01-06:
Saran anda sudah saya lakukan tapi eror tetap ada

nopi890, 2024-01-06:
[image]

RomanLut, 2024-01-26:
I am sorry this error is not related to hx_espnow_rc project, but to PlatformIO installation. I can not help with that.

closed 2024-01-26 by RomanLut