RomanLut/hx_espnow_rc #16: Tx and Rx Communication

devanesanandrews, 2024-01-10:
Hi Thank you for a wonderful project. I am using 2 esp32 in LR mode.

When i use the default Profile 1, In the Trasnmitter serial monitor i am getting this

Receiver    | FS: 1 | RSSI: 0 | Recv/retransm: 0/0( 0%)) | Lost: 0 | Invalid/CRC: 0/0 | Tel. overflow/resync: 0/0 | In telemetry:@0 D/s
!��,��time:31
T�Xk���ѕ�� | FS: 1 | RSSI: 0 | Total: 1177 | Ack: 0 | Error: 0 | Missed time: 30 | ACKed ��I��%� 0p/s | Out telemetry: 0 b/s
            | RSSIDBm: -0dbm | Noise Floor: -0db� | SNR: 0db | WifiRatet -1
Receiver    | FS: 1 | RSSI: 0 | Recv/retransm: 0/0(0%)) | Los�� 0 | Invalid/CRC: 0/0 | Tel. overflow/resync: 0/0 | In �elemetry: 0 b/s
!�����time:31
T�Xk���ѕ�� | FS: q | RSSI: 0 | Total�����>�Ack: 0 | Error: 0 | Missed time: 31 | ACKed PktRate: 0p/s | Out telemetry: 0 b/s
    @       | RS�IDBm: -0db� | Noise Floor: -0dbm | SNR: 0db | WifiRate: -1
Receiver �  | FS: 1 | RSSI: 0 | Recv/retransm: 0/0( 0%)) | Lost: 0 | Invali�/CRC: 0/0 | Tel. overfHow/resync:`0/0 | In telemetry: 0 b/s
!�����time:32
Transmitter | FS: a | RSSI: 0 | Total: 1275 | Ack: 0 | Error: 0 | Missed time: 32 | ACKed PktRate: 0p/s | Out telemetry: 0 b/s
            | RSSIDBm: -0dbm | Noise Floor: -0dbm | SNR: 0db | WifiRate: -1
Receiver    | FS: 1 | RSSI: 0 | Recv/retransm: 0/0( 0K)) | Lost: 0 | Invalid/CRC: 0/0 | Tel. overflow/resync: 0/0 | In telemetry: 0 b/s
!�����time:32
T�Xk���ѕ�� | FS: 1 | RSSI: 0 | Total: 1324 | Ack: 0 | Error: 0 | Missed time: 33 | ACKed PktRate: 0p/s | Out telemetr�: 0 b/s
            | RSSIDBm: -0dbm | Noise Floor: -0dbm | SNR: 0db | WifiRate: -1
Receiver    | FS: 1 | RSSI: 0 | Recv/retransm: 0/0( 0%)) | Lost: 0 | In���J��CRC: 0/0 | Tel. overflow/resyn�: 0/0 | In telemetry: ` b/s
a�����time:31

I am using ESP32 with PPM bidirectional telemetry, is there any special configuration needed in the profile for transparent telemetry?

Kindly help

RomanLut, 2024-01-10:
"FS:1"means there is no communication between modules.

Corrupted characters are normal because  software UART emuation used (no free HW UARTS left).

1) Try to capture debug log right after reset. You should see something like:

```
Got profile:0
Starting profile 0      
/profile1.json
Waiting for CH16 profile
HXESPNOWRC: Inf�: Board MAC�X␖���͡STA): 7C:9G:BA:F4:C2:28 
HXESPNOW�C: Info: Bocrd MAC address(AP): 7C:8E:BD:F5:C2:29
G��profile:0
!Cycle time:18373
Transmitter | FS: 1 | RSSI: 0 | Total: 50 | Ack: 0 | Error: 0 | Missed time: 8 | ACKed PktRate: 0p/s | Out te�emetry: 0 b/s
   @        | R�SIDBm: -0dbm | Noise Floor: -0dbm | SNR: 0db | WifiRate: -1

```   
   
2) Make sure receiver project is compiled with the same settings as specified in the profile1.json:
```
	"espnow_channel": 3,
	"espnow_key": 0,
	"espnow_long_range_mode": false,

```
There should be FS:0 on both sides even if you do not send any telemetry.

devanesanandrews, 2024-01-10:
hey Thank you, it worked, But i found some issues.

I am using ESP32 as both TX and RX

Case 1: *Works without any issue, led blinks in esp32 developement board*
Tx - Espnow LR mode is false
Rx - Espnow LR mode is false

Case 2: there is no communication, but after the time limit, RX doesnt go into the AP mode
Tx - Espnow LR mode is false
Rx - Espnow LR mode is true

Case 3: There is no communication
Tx - Espnow LR mode is true
Rx - Espnow LR mode is true

Case 3: There is no communication
Tx - Espnow LR mode is true
Rx - Espnow LR mode is false

RomanLut, 2024-01-12:
Thank you for bug report.
Threre was indeed bug added recenlty: LR mode setting was not read correctly from profile json.
Fixed on master: https://github.com/RomanLut/hx_espnow_rc/commit/9c6a12222aa7e4af4c202dcc46fa879a4a815237

devanesanandrews, 2024-01-14:
@RomanLut now its working.

one more doubt, in the RX the telemetry baudrate is set in the rx.config file.

But in the Tx file its is not there, when i checked the code, Hc06 software serail begin is in 115200 baudrate. 

when i tested with Mavlink telemetry is not flowing

RomanLut, 2024-01-14:
You have to reconfigure HC06 module to 115200 baudrate before connecting to tx module. If it still does not work, I would suggest testing using usb-uart convertor connected to tx. 
If debug output shows non-zero telemetry rate, then telemetry is transmitted.

devanesanandrews, 2024-01-15:
@RomanLut, Thank you for your support, its working very well.

Transmitter | FS: 0 | RSSI: 96 | Total: 4212 | Ack: 3985 | Error: 0 | Missed ti�e: 95 | ACKed PktRate: 48p/s | Out telemetryz 0 b/s
            | R�SIDBm: -25dbm | Noise Floor: -96dbm | SNR: 71db | WifiRate: 25
Receiver    | FS: 0 ��RSSI: 100 | Recv/retransm: 4008/24( 0%i) | Lost: 0 | Invalid/CRC: 0/0 | �el. overflo�/resync: 0/1 | In tel�metry: 20 b/s
!Cycle time:32
Transmitter | FS: 0 | RSSI: 98 | Total: 4261 | Ack: 4034 | Error: 0 | Missed time: 96 | ACKed PktRate: 49p/s | Out telemetry: 0 b/s
            | RSSIDBm: -26dbm | Noise Floor: -96dbm | SNR: 70db | Wif�Rate: 25
Receiver    | FS: 0 | RSSI: 100 | Recv/retrans�: 4057/24( 0%)) | Lost: 0 | Invalid/CRC: 0/0 | Tel. overflow/resync: 0/1 | In t�lemetry: 20 �s
!Cycle time:32
Transmitter | FS: 0 | RSSI: 98 | Total: 4310 | Ack: 4083 | Error: 0 | Missed tim�: 97 | ACK�d PktRate: 49p/s | Ou��telemetry: 0 b/s
            | RSSIDBm: -25dbm | Noise �loor: -96db� | SNR: 71db | WifiRa�e: 25
Receiver    | FS: 0 | RSSI: 100 | Recv/retransm: 4106/24( 0%)) | Lost: 0 | Invalid/CRC: 0/0 | Tel. overflo�/resync: 0/1 | In tele�etry: 20 b/s

But my telemetry rate is only 20b/s, i have connected to mavlink

closed 2024-01-21 by devanesanandrews