RomanLut/hx_espnow_rc #17: 10 Position switch for Profile Selector - *TX EXternal Module* - Enhancement

devanesanandrews, 2024-01-21:
Hi thank you for the project. 

I am using a diy transmitter with sbus output with no display. Its kind of difficult to control the profile selections.

Can we have a profile selection switch in the external module. Like a 10 position switch like this or 

[image]

a 4 way position switch

[image]

RomanLut, 2024-01-22:
>Can we have a profile selection switch in the external module?

You will have to implement it yourself. 
I have no problem setting channel output to constant value with EdgeTX.

>Also i found one more issue, everytime you edit a profile, the profile goes to the last place and a diffirrent profile gets selected for the first selection example 1050. 

I did not experienced this issue. You have to double check what profileN.json is loaded in the debug output .

>Whats your feedback on using Tx External Module and RX esp32 Sbus in LR mode, is it feasible for RC?

RC system works very well for me at distance up to 1.5Km. LR mode adds ~20% to the range.
I am not drone racer thought; I am using it on micro RC planes, so experience may be different for you if you expect high packet rates (update rate is ~40Hz).

devanesanandrews, 2024-01-22:
@RomanLut  Thank you for your insights

You will have to implement it yourself.
- I am working on it, would you give me a hint or place where i should look into it.
I am using ppm so 40Hz should be more than enough

I did not experienced this issue. You have to double check what profileN.json is loaded in the debug output 
- is the profiles and values fixed?

RomanLut, 2024-01-26:
> * I am working on it, would you give me a hint or place where i should look into it.

Assign s a profile index which you read from switch here:
https://github.com/RomanLut/hx_espnow_rc/blob/9c6a12222aa7e4af4c202dcc46fa879a4a815237/lib/hx_rc_transmitter_common/modeBase.cpp#L46C17-L46C48