RomanLut/hx_espnow_rc #18: Esp32cam mavlink rc

nopi890, 2024-01-25:
Saya flash esp 32cam dengan contoh RXesp32cam mavlink ,berhasil terhubung dengan DIYtxcontroller tapi sayangnya tidak ada tampilan gambar modul esp32cam LED putih berkedip cepat ,apakah harus menambahkan kartu memori agar esp32cam bisa mengirim gambar,

AutoPlantBali, 2024-01-25:
> Saya flash esp 32cam dengan contoh RXesp32cam mavlink ,berhasil terhubung dengan DIYtxcontroller tapi sayangnya tidak ada tampilan gambar modul esp32cam LED putih berkedip cepat ,apakah harus menambahkan kartu memori agar esp32cam bisa mengirim gambar,

comment pake bahasa inggris bro.... sertakan screenshoot di serial monitor atau video pendek

nopi890, 2024-01-26:
Nanti juga ditranslate sama google ,kemaren2 juga dibales, ngapa situ yang
repot, heran

AutoPlantBali, 2024-01-26:
@nopi890 
jajaja....lol...
discussing without ethics and manners, This is not an issue but you don't read enough literacy, I hope this issue is closed

RomanLut, 2024-01-26:
rx_esp32cam_mavlink_rc.code-workspace is in proof-of-concept state, it is not finished, and work is suspended due to luck of inteterest. You can use source code as starting point for your own project only.

closed 2024-01-26 by RomanLut