RomanLut/hx_espnow_rc #21: TX External Module SBUS and receiver

AutoPlantBali, 2024-03-15:
hi.. @RomanLut 

I use a Jumper T12 radio and SBUS output if the TX External Module is turned on with external power and connected to the receiver the SBUS output is normal but if it is turned on at the same time the radio and TX module cannot be connected to the receiver, can it be set if the TX module is connected to the receiver then it receives the signal sbus from radio?

RomanLut, 2024-03-16:
Sorry I do not understand your question.
In any case, the first thing you should do is to examine debug output from TX module.

AutoPlantBali, 2024-03-16:
result debug sbusDecorder.dump() if the radio is turned on together with the tx module:

--- Terminal on COM7 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
rst:0x1 (POWERON_RESET)␀Waiting for CH�␂�ɽ����k
!Cycle time:2830
Got profile:5
!Cycle time:�0
!Cycle time:46
!Cycle time:46
!Cycle time:46
!Cycle time:46
!Cycle time:46
!Cycle time:46
!Cycle time:46
!Cycle time:46
!C�cle time:46
!Cycle time:46
!Cycle time:46
!C{cle time:46
!Cycle time:46
!Cycle time:46
!Cycle time:46
!C{cle time:46
!Cycle time:46
!Cycle time:46
!Cycle time:4v
!Cycle time:46
!Cycle time:46
!Cycle time:46
!Cycle time:46
!Cycle time:46
!Cycle time:46
!Cycle time:46
!Cycle time:46
!Cycle time:46

===========================
result debug sbusDecorder.dump() if the radio is turned on after the tx module is connected to the receiver:

--- Terminal on COM7 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H

clk_drv:0x␀Waiting fnr CH16 profile
!Cycle time:2825
Got profile:5
!Cycle time:45
!Cycle time:42
Starting prof�le index=5
/profile6.json
␔LR Mode=f�lse16 profile
HXESPNOWRCr Info: Boa�d MAC addre�s(STA): EC:62:60:A7220:E8
␖Got profile:5fo: Board MAC address(AP): EC:62:60:A7:20:E9
␖Event:CONNECTED
!Cycle`time:43
Transmitter␀| FS: 0 | �SSI: 63 | �otal: 48 | Ack: 39 | Error: 0 | Missed time:`9 | ACKed PktRate: 31p/s | Out te�emetry: 153 b/s
            | RSSIDBm: -52dbm | Noise Floor: -97dbm | SNR: 45db | WifiRater 0
Receiver    | FS: 0 | RSSI: 25 | Recv/retransm: 48/10(25%)) | Lost: 0 | Invalid/CRC: 0/0 | Tel. overflow/resync: 0/1 | I� telemetry: 0 b/s
SBUS FS!
!Cycle time:32
!Cycle time:42
Transmitter | FS: 0 | RSWI: 93 | Total: 95 | Ack: 85 | Error: 0 | Missed time: 11 | ACKed Pk�Rate: 46p/s | Out tele������ 75 b/s
        `   | RSSIDB�: -52dbm | Noise Floo�: -97dbm | SNR: 45db | WifiRate: 0
Receiver  � | FS: 0 | RSSI: 43 | Recv/retransm: 95/10( 0%)) | Lost: 0 | Inval�d/CRC: 0/0 | Tel. overflow/resync: 0/1 | In telemetry: 0 b/s
SBUS FS#
!Cycle time:31
!Cycle time:43
Tra�smitter | FS: 0 | RSSI: 97 | Total: 142 | A�k: 133 | Error: 0 | M�ssed time: 13 | ACKed �ktRat�� 48p/s | Out telemetry: 73 b/s
            | RSSI@Bm: -52dbm | Noise Floor: -97dbm | SNR: 45db␀| WifiRate: 0
Receiver    | FS: 0 | RSSI: 58 | Recv/retransm: 142/10( 0%)) | L�st: 0 | Invalid/CRC: 0/0 | Tel. overflow/resync: 0/1 | In telee 
try: 0 b/s
SBUS FS!
!Cycle time:33␙
!Cycle tim�:43
Transmitter | FS: 0 | R�SI: 95 | Total: 189 | Ack: 180 | Error: 0 | Missed time: 15 | ACKed PktRate: 47p/s | Out telemetry: 75 b/s
            | RSSIDBm: -52dbm | Noise Floor: -97dbm | SNR: 45db | WifiRate: 0
Receive�    | FS: 0 | RSSI: 68 | Recv/retransm: 189/10( 0%)) | �ost: 0 | Invalid/CRC: 0/0 | Tel. �ٕə��ݽres�nc: 0/1 | �n telemetry
: 0 b/s
SBUS FS!
!Cycle time:3�C�!Cycle time:42

AutoPlantBali, 2024-03-16:
[image]

[image]

RomanLut, 2024-03-16:
I am not sure where the problem is.
TX module should be in the ModeIdle::Loop() after start: https://github.com/RomanLut/hx_espnow_rc/blob/06e7f90bead54aaa519656b3936a0d24ebfe12d6/lib/hx_rc_transmitter_common/modeIdle.cpp#L22

I see that module detects profile 5:

```Got profile:5```

Here: https://github.com/RomanLut/hx_espnow_rc/blob/06e7f90bead54aaa519656b3936a0d24ebfe12d6/lib/hx_rc_transmitter_common/modeBase.cpp#L50

And it has to start profile 5 after 1 second: https://github.com/RomanLut/hx_espnow_rc/blob/06e7f90bead54aaa519656b3936a0d24ebfe12d6/lib/hx_rc_transmitter_common/modeBase.cpp#L86

But it does not happend, there is no ```Starting prof�le index=5``` line like in the second log.

I suggest to add ```HXRCLOG.println()``` statements to ```ModeBase::haveToChangeProfile()``` to find out why it does not return ```true```.

AutoPlantBali, 2024-03-16:
add haveToChangeProfile() this line:

void ModeBase::loop(
  const HXChannels* channels,
  HC06Interface* externalBTSerial,
  Smartport* sport
)
{
    // HXRCLOG.print("failsafe:");
    // HXRCLOG.println(channels->isFailsafe);
    if (channels->isFailsafe)
    {
        int s = getProfileIndexFromChannelValue(channels->channelValue[15]);

        if ( this->CH16ProfileIndex != s )
        {
            HXRCLOG.print("Got profile:");
            HXRCLOG.println(s);
            HXRCLOG.println(haveToChangeProfile());

            this->CH16ProfileIndex = s;
            this->gotCH16ProfileTime = millis();
        }
    }

result debug print:
␀��ѥ���␀�K�CH16 profile
!Cycle time:346
Got profile:k
0
!Cycle time:46
!Cycle time:46
!Cycle time:46
!Cycle time:47
!Cycle time:46
!Cycle time:46
!Cycle time:46
!Cycle time:46
!Cycle time:46

RomanLut, 2024-03-16:
There is no sense to call ```haveToChangeProfile()```  while ```gotCH16ProfileTime``` is still zero, it will return "false" (0) for sure.

You have to print all statements used inside ```ModeBase::haveStableCH16ProfileSelection()``` to find out why it returs false.

AutoPlantBali, 2024-03-16:
Thank you very much @RomanLut 

I have found the cause, after I restored the git commit. this is because of missing "!" in code:

void ModeBase::loop(
   const HXChannels* channels,
   HC06Interface* externalBTSerial,
   Smartport* sport
)
{
     if ( !channels->isFailsafe)
     {

Now it is running normally with SBUS and PPM output

closed 2024-03-16 by AutoPlantBali