RomanLut/hx_espnow_rc #23: old 9x TX with js bay

pepelillo, 2024-04-01:
Hello. I want to try this project in my old turnigy 9x with opentx (last version supported) and make al diy js module for this TX. Is this possible? not sure if I can set the TX in sbus mode.....maybe PPM.

RomanLut, 2024-04-01:
If you can select "External RF" "Mode:SBUS" on SETUP page, it should work.

pepelillo, 2024-04-01:
Thanks for the fast answer. Unfortunately I can not see this option in radio setup....opentx versión 2.1.1. I can only see in modelo setup an option "mode" with ppm ppm16 and ppmsim. Any ideas?

geofrancis, 2024-04-01:
I dont think the atmega chip that the 9x uses is capable of inverted serial needed for SBUS.

pepelillo, 2024-04-01:
So, not posible? 

Anyway congrats for this great project. I wish It can Accept ppm signals as input to give a second live to mu old turnigy....

geofrancis, 2024-04-01:
you could add a ppm to sbus converter between the radio and the esp32.

pepelillo, 2024-04-01:
Great.....i Will try.
Ana the rest is It similar to the guides in the micro Jr module?
Can you guide me somehow?
Thanks