RomanLut/hx_espnow_rc #24: enhancement Bluetooth gamepad to ppm or sbus

Pepeuch, 2024-09-02:
Hello, I see your project is quite successful but I am looking for something that apparently you have not yet achieved.
I explain to you:
- I have a rover or ardurover, my transmitters are a little old and I don't need long distance, I'm looking for a way to use a gamepad (sixaxis or dualshock4) which transmits via bluetooth.
- The rover would receive the bluetooth signals emitted by the gamepad and convert it into a ppm or sbus signal on the flight controller. do you think this is possible with an esp32 or d1 mini? see even an esp32 c3 which are now very compact.

RomanLut, 2024-09-02:
Yes it is possible and has been used in another project already:
https://github.com/RomanLut/Multiwii-ESP32-port?tab=readme-ov-file#ps3-controler

via https://github.com/jvpernis/esp32-ps3
or https://github.com/aed3/PS4-esp32

however you are on your own to implement esp32 receiver which connects to dualshock and outputs sbus; this fucntionality is not related to this project.

Pepeuch, 2024-09-04:
good evening, thank you for your response, I have already come across this library, it works very well, I am in the process of making a bluetooth receiver based on esp32 d1mini I would not see the continuation for an esp32 s3 supermini with modified antenna, but the goal is to be able to output a ppm signal compatible with ardupilot (I use your hx_PPM_encoder library). I hope this works I will keep you posted.

https://github.com/RomanLut/Multiwii-ESP32-port?tab=readme-ov-file#ps3-controler